stm32-rs/stm32h5xx-hal #1: Current Status/Plan?

astapleton, 2023-07-21:
Hi there! I'm looking to start work with an STM32H503 in the near future and was wondering what the status or plan for the development of the STM32H5xx HAL is. Is there something holding it back or do you just need contributions/time?

Dirbaio, 2023-08-25:
for anyone wanting to use Rust on STM32H5, [`embassy-stm32`](https://github.com/embassy-rs/embassy) has fairly complete support, see [examples](https://github.com/embassy-rs/embassy/tree/main/examples/stm32h5/src/bin).

olback, 2023-10-09:
Hi :wave: 

My plan was to work on this on my free time (evenings and weekends). Unfortunately, work got incredibly busy and I'm working many hours overtime every day and will for at least another month. 

If anyone else is interested in working on this, please do! If anyone picks up this project, I will try to contribute as much as I can when work calms down a bit.

 I think @adamgreig (sorry if I got the wrong person here) should have permissions to add people to the project.

Sorry for taking so long to reply to this.

adamgreig, 2023-10-09:
Hi @olback, thanks for the update!

Would you be happy to add the stm32h5 team as an owner to the stm32h5xx-hal crate, so it's possible for others to eventually make releases too? If so you should be able to run `cargo owner --add github:stm32-rs:stm32h5` from the `stm32h5xx-hal` directory. Currently you're the only team member, but it's useful so that if you're not able to, other future team members could make a release.

olback, 2023-10-10:
@adamgreig 

I've added the stm32h5 team now!

astapleton, 2023-10-11:
Thanks for the update @olback! Appreciate you working with @adamgreig on this!

closed 2024-05-31 by astapleton