stm32-rs/stm32h5xx-hal #35: Feature request - ADC

usbalbin, 2025-01-20:
Add support for the ADCx peripherals which are available on all devices, also including H523 and H533.

usbalbin, 2025-01-21:
No problem, sounds good! I am happy to give this a go in (hopefully not too distant) future, unless someone else gets to this first.

Thanks :)

BTW, anyone know if H5's ADC's are similar to that of any other family?

astapleton, 2025-01-21:
Looks like a number of processors share the same ADC peripheral (see [here](https://github.com/search?q=repo%3Astm32-rs%2Fstm32-rs%20fields%2Fadc%2Fadc_v3.yaml&type=code)). Notably, the H7 series does, and I've based a lot of my work on [stm32h7xx-hal]https://github.com/stm32-rs/stm32h7xx-hal (although I've demacro-fied some of the drivers, taking inspiration from [stm32f4xx-hal]https://github.com/stm32-rs/stm32f4xx-hal in order to do so). See the I2C PR (#27) for an example (ie. Instance trait, and usage of generics).