stm32-rs/stm32h5xx-hal #36: Feature request - SPI

Wratox, 2025-01-22:
Add support for the SPIx peripherals.

Might try to add this myself, but is there any reason to not go straight for embedded_hal v1.0.0? 
If so do you think [https://github.com/stm32-rs/stm32h7xx-hal/pull/476](https://github.com/stm32-rs/stm32h7xx-hal/pull/476) is a good place to start?

astapleton, 2025-01-22:
@Wratox I have a SPI implementation that will implement Master + Slave + embedded-hal 1.0.0 that I will start putting up for review soon!

Wratox, 2025-01-23:
@astapleton Alright, nice!

I did start on it and I did the pins for SPI for stm32h573, is that something you have a use for or have you already implemented that?

astapleton, 2025-01-23:
@Wratox yes, that would be helpful. I don't have pins for anything other than the STM32H503 right now

Wratox, 2025-01-23:
@astapleton [It's available at spi.rs at my spi branch.](https://github.com/Wratox/stm32h5xx-hal/blob/9f7ea47f37db5d9a8ad6bf61a3b82ec19c958b35/src/spi.rs#L344-L536)
I also did the pins for stm32h523, there's a comment in the code about what datasheets I've mapped.

tytouf, 2025-05-02:
@Wratox any progress on your side ?

I'm working on a board with an stm32h533 and would really like to use your code if it is available. Of course if it needs any testing or changes I can also contribute.

Thanks,

astapleton, 2025-05-02:
@tytouf I've been working on the SPI when I have time. The latest version I've pushed is on https://github.com/stm32-rs/stm32h5xx-hal/tree/as/spi-master. I've made some changes since I published that, and I need to just find some time to test that it still works, but if you stick to the `embedded-hal` traits it should be good when it merges. I think once that is merged I can do some cleanup and publish to crates.io too.

tytouf, 2025-05-20:
Thanks @astapleton I'll take a look.

astapleton, 2025-07-10:
Reopened because I still need to put up the slave operation PR!