stm32-rs/stm32h5xx-hal #39: Add support for STM32H523/STM32H533

astapleton, 2025-01-23:
This is a tracking issue for the changes required to add support for the STM32H523/STM32H533 group of processors to the existing functionality.

- [ ] Integrate `stm32h5` PAC with STM32H523/STM32H533 support (need new stm32h5-staging release, also see https://github.com/stm32-rs/stm32-rs/issues/1122)
- [ ] Define GPIOs in GPIO driver
- [ ] Define I2C pins in I2C driver (WIP - see https://github.com/stm32-rs/stm32h5xx-hal/pull/27)
- [ ] Define SPI pins in SPI driver (WIP)

usbalbin, 2025-01-24:
Aside from any potential bugs, H523 and H533 support has now been merged in the pac

astapleton, 2025-01-24:
> Aside from any potential bugs, H523 and H533 support has now been merged in the pac

We also need the stm32h5-staging crate to be updated to bring in those changes (instead of pointing the HAL at the nightlies and accidentally bringing in new breaking changes). @burrbull is this something you could do now or should we wait for a wider stm32-rs update?

astapleton, 2025-07-15:
I think this can be considered done now. All the implemented drivers support STM32H503/523/533/562/563/573