stm32-rs/stm32h5xx-hal #49: Peripheral overview

usbalbin, 2025-05-27:
| Done | Peripheral/Feature | Issue | PR |
|---------|-----------------------------|----------|-----|
| [ ]       | ADC                           |    #35 |    - |
| [ ]       | CAN                           |         ? |@oshaughnessya WIP?|
| [ ]       | Timers                      |          - |    - |
| [ ]       | PWM                         |          - |    - |
| [ ]       | Rng                           |     #34|#37|
| [ ]       | I2C                            |          - |#45|
| [ ]       | SPI                            |    #36 |@astapleton / @Wratox WIP|
| [ ]       | UART                        |          - |    - |
| [ ]       | Flash                         |         - |    - |

Feel free to edit and/or add things

astapleton, 2025-05-27:
Thanks for putting this together.

I have a SPI implementation following the guidance set up in the I2C reviews that is almost ready to go, although my refactoring broke something so I just need to find the time to fix that.
I also have a working UART implementation (implements embedded-hal serial traits), but it needs a similar treatment to the SPI, and I also just need to find a bit of time to tackle that. 

I realize I might be blocking someone, so I can potentially work with someone to take over the UART work, and carve out some time to finish the SPI (I think it would be easier for me to just to finish that as I have the context there)

astapleton, 2025-05-27:
I've also got a DMA implementation that I need to update with some changes to the PAC that are now available as of 0.16.0 before it can be published.

astapleton, 2025-06-10:
No rush. Happy for you to take your time. I've generally also tried to use traits + generics to reduce the reliance on macros like in the H7 repo. It's worked decently well without much modification to the overall structure of the functions implemented by those macros.

astapleton, 2025-06-10:
I'm also happy to test an example on an H503 to confirm the driver works.

tytouf, 2025-06-11:
> I'm also happy to test an example on an H503 to confirm the driver works.

Yes.

@astapleton back to the SPI, did you succeed in using the SPI Slave mode ? I simply modified the code from your PR to disable the master mode and I can't seem to trigger any transmission back from the STM32h533 even with no hardware management of the slave select pin. Tx buffer is loaded but status in unchanged even though I push transfers from an Arduino board.

tytouf, 2025-06-12:
Thank you @astapleton 

I pushed a branch with a pull request for the timer support: https://github.com/stm32-rs/stm32h5xx-hal/pull/59

phansel, 2025-12-02:
is there any interest in UART/USART? hard to believe this is still open. everyone must be using defmt?

astapleton, 2025-12-03:
> is there any interest in UART/USART? hard to believe this is still open. everyone must be using defmt?

As mentioned above, I've got an implementation that needs some work to get over the line. I've just had very little time to dedicate to doing so over the last few months. I'd be happy to help someone take what I have and complete it. Not sure what defmt has to do with it though?

astapleton, 2026-01-08:
@phansel PR up for U(S)ART asynchronous serial functionality: #75